fprime-community/fprime-arduino #1: fprime-util install fails for libSvc_LinuxTime

brinkap, 2020-11-01:
## Issue
I'm able to successfully ``` fprime-util generate``` and ```fprime-util build``` the application. When I attempt to run ```fprime-util install``` I get the following CMake error:

``` bash
CMake Error at F-Prime/Svc/LinuxTime/cmake_install.cmake:36 (file):
  file INSTALL cannot find
  "/home/dev/oosfprime/fprime-arduino/ArduinoBlink/build-fprime-automatic-Teensyduino/lib/Teensyduino/libSvc_LinuxTime.a":
  No such file or directory.
Call Stack (most recent call first):
  F-Prime/Svc/cmake_install.cmake:60 (include)
  cmake_install.cmake:39 (include)
```

## Workaround
I have a work around to go into the fprime/Svc/CmakeLists.txt and fprime/Drv/CmakeLists.txt and comment out the offending components like so:

``` cmake
# in fprime/Svc/CmakeLists.txt
...
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/GroundInterface/")
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/Health/")
# add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/LinuxTime/")
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/LinuxTimer/")
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/PassiveConsoleTextLogger/")
...
```
``` cmake 
# in fprime/Drv/CmakeLists.txt
...
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/LinuxSpiDriver/")
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/LinuxI2cDriver/")
# add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/SocketIpDriver/")
```
Then the application builds as expected.

## More Info

Upon closer inspection, this is the first of two fprime libraries excluded by the application in ```fprime-arduino.cmake```
``` cmake
set_target_properties(
    Svc_LinuxTime
    Drv_SocketIpDriver
    PROPERTIES EXCLUDE_FROM_ALL 1 EXCLUDE_FROM_DEFAULT_BUILD 1)
```
It appears the target propery is being correctly set and the static libraries are in fact properly excluded from the build. However, the install step is still looking for the library and is not properly excluded from the build so to speak. 

The ```fprime-util generate``` step invokes the ```generate_library``` CMake macro (defined in ```cmake/support/Module.cmake```) 
``` cmake
function(generate_library MODULE_NAME SOURCE_FILES_INPUT DEPS_INPUT)
  # ... (code omitted)
  # Install the executable, if not excluded and not testing
  get_target_property(IS_EXCLUDE_FROM_ALL "${MODULE_NAME}" "EXCLUDE_FROM_ALL")
  if ("${IS_EXCLUDE_FROM_ALL}" STREQUAL "IS_EXCLUDE_FROM_ALL-NOTFOUND" AND
      NOT CMAKE_BUILD_TYPE STREQUAL "TESTING") 
      install(TARGETS "${MODULE_NAME}"
          RUNTIME DESTINATION "bin/${PLATFORM}"
          LIBRARY DESTINATION "lib/${PLATFORM}"
          ARCHIVE DESTINATION "lib/static/${PLATFORM}"
      )
  endif()
  # ... (code omitted)
endfunction(generate_library)
```
But the target property cannot be set before invoking this support macro since it does not yet exist. 

I imagine with the new out-of-source build in Fprime 1.5.1 that a proper solution would be to not modify the source fprime code in any way. Perhaps there could be a new fprime_* macro that could be used to properly exclude default Svc and Drv components or perhaps the execution order of the fprime-util could be modified so the the install logic checking for the EXCLUDE_FROM_ALL property would occur after the application had a chance to set that property (likely in a second pass after the ```generate_library``` macro is called.)

Great work on the out-of-source build and user documentation! Great work!

Joshua-Anderson, 2020-11-02:
@brinkap Thanks so much for investigating this! This writeup was super awesome and helpfu. We rebuilt the cmake install process in our internal repo because we wanted to avoid building unnecessary libraries, but I didn't consider the implications for building excluded libraries, so that's a good find.

I think our install target refactor should resolve this issue, and I'll make sure to double check against the arduino target.

brinkap, 2020-11-13:
@Joshua-Anderson Thanks for response. Do you have a FPrime target release for that target refactor that should fix this issue? I'm interested in this feature not just for a this Arduino example but for a TI Hercules cortex-R5 board I've been working on in support of a JPL application. If internal development branches implement this change, I test it out for you too.

Joshua-Anderson, 2020-11-13:
The last part of the changes were merged in [here](https://github.com/nasa/fprime/pull/242). It'll be released in the next fprime release, but you can test it on the devel branch now.

I haven't tested the changes against this arduino project yet since I don't have toolchain setup (it'll get tested sometime before the offical release), but I'm pretty sure the issue is resolved now.