fprime-community/fprime-arduino #2: How to install on Arduino?

Ruben1729, 2021-01-09:
| | |
|:---|:---|
|**_Fprime Commit_**|[20b5f29](https://github.com/nasa/fprime/tree/20b5f2950c5f264a1be3d47e0e94a69518d7aeca)|
---
## Question About Installation

I was wondering if you could provide further instructions on how to install the project on an Arduino. I have run the `fprime-util generate` and `fprime-util build` commands successfully but I'm unsure where and what binaries are needed to run the project on the Arduino. 

The commands generate the `build-fprime-automatic-Teensyduino` folder in which I have found that the `lib/Teensyduino` folder contains static libraries. Are those the binaries needed to run the project on the Arduino? If so, how does one go about installing it on the microcontroller? Additionally, which pins should I connect the LED to for the project to work?

Thank you!

Joshua-Anderson, 2021-01-11:
The Arduino example hasn't been updated for some of the cmake changes we've made to the development fprime branch yet. Our example projects are typically pinned to the latest F' release, so if you use the version of F' the submodule of this project points to (F' version 1.5) it should build alright.

If there's a particular reason you need to use development version of F' on Arduino we can probably help out and track down what some of the issues, but generally I'd recommend using the latest F' release.

We usually update the example F' projects with each F' release, so we'll normally go through and fix all these issues when we're making the next official F' release off the F' development branch.

Ruben1729, 2021-01-11:
Awesome, I gave it a shot and it built the binaries properly. Now I have the issue that the ROM size is not big enough on the Arduino UNO but that's a problem on my side. I will be posting an image for future devs looking at the project issues!

[image]

Closing the issue.

Thank you for everything!

closed 2021-01-11 by Ruben1729