fprime-community/fprime-arduino #20: CMake Error - The dependency target "Svc_Time" of target "Arduino_ArduinoTime" does not exists.

amalnathh, 2023-12-11:
After the Fprime v3.4.0 the arduino lib failed to generate for all boards. 
See: [Convert Svc.Time into An Interface by @LeStarch](https://github.com/nasa/fprime/pull/2317) 

I think it may be due to the changes in Svc/Time.
This is the Error
> CMake Error at fprime/cmake/target/build.cmake:83 (add_dependencies):
> --Generating done
>    The dependency target "Svc_Time" of target "Arduino_ArduinoTime" does 
>    not exists.
> Call Stack (most recent call first):
>         fprime/cmake/target/build.cmake:141 (build_setup_build_module)
>         fprime/cmake/target/target.cmake:141 (build_add_module_target)
>         fprime/cmake/target/target.cmake:116 (cmake_language)
>         fprime/cmake/target/target.cmake:155 (setup_single_target)
>         fprime/cmake/module.cmake:48 (setup_single_target)
>         fprime/cmake/module.cmake:88 (generate_base_module_properties)
>         fprime/cmake/API.cmake:228 (generate_library)
>         fprime-arduino/Arduino/ArduinoTime/CMakeLists.txt:19 (register_fprime_module)

Steps to recreate:
Start a fresh venv and follow the arduino tutorial in the [fprime website](https://fprime-community.github.io/fprime-tutorial-arduino-blinker/)
follow the arduino cli guide for any board, add fprime-arduino as a lib, and then fprime-util generate.

ethancheez, 2024-01-09:
My [PR](https://github.com/fprime-community/fprime-arduino/pull/19) addresses this issue.

You can also try checking out the [`fprime3.4.0`](https://github.com/fprime-community/fprime-arduino/tree/fprime3.4.0) branch of `fprime-arduino` to test if it fixes the issue.

ethancheez, 2024-01-20:
PR was merged, this cmake error should be fixed. Could you test on your end?

amalnathh, 2024-01-20:
Tested on my end. Working properly with the fprime3.4 branch. Cheers!!

closed 2024-01-20 by amalnathh