fprime-community/fprime-arduino #27: Generate not working for RP2040

csmith608, 2024-03-31:
Hello, 

I've been trying to run the arduino led blinker tutorial on the RP2040. https://fprime-community.github.io/fprime-tutorial-arduino-blinker/ 

I'm getting stuck on the first `fprime-util generate`. I'm getting the error shown below 

```
[INFO] Generating build directory at: /Users/anniesmith/led-blinker/build-fprime-automatic-rpipicow
[INFO] Using toolchain file /Users/anniesmith/led-blinker/fprime-arduino/cmake/toolchain/rpipicow.cmake for platform default
Traceback (most recent call last):
  File "/Users/anniesmith/fprime-venv/bin/arduino-cli-wrapper", line 5, in <module>
    from arduino_cli_cmake_wrapper.cli import main
  File "/Users/anniesmith/fprime-venv/lib/python3.8/site-packages/arduino_cli_cmake_wrapper/cli.py", line 193, in <module>
    def assemble_output_data(test_files: dict[Source, Path], stages: Dict[Stage, List[str]], detect: bool, include: bool, post_link: bool) -> Tuple[Dict, Path]:
TypeError: 'type' object is not subscriptable
CMake Error at fprime-arduino/cmake/toolchain/support/arduino-wrapper.cmake:72 (message):
  Could not run 'arduino-cli' please install 'arduino-cli' and board
  'rp2040:rp2040:rpipicow'.
Call Stack (most recent call first):
  fprime-arduino/cmake/toolchain/support/arduino-support.cmake:43 (run_arduino_wrapper)
  fprime-arduino/cmake/toolchain/support/arduino-support.cmake:201 (set_arduino_build_settings)
  fprime-arduino/cmake/toolchain/rpipicow.cmake:19 (include)
  /usr/local/Cellar/cmake/3.29.0/share/cmake/Modules/CMakeDetermineSystem.cmake:146 (include)
  CMakeLists.txt:7 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
-- Configuring incomplete, errors occurred!
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
```

Here's my settings.ini file 

```
[fprime]
project_root: .
framework_path: ./fprime

default_cmake_options:  FPRIME_ENABLE_FRAMEWORK_UTS=OFF
                        FPRIME_ENABLE_AUTOCODER_UTS=OFF
library_locations: ./fprime-arduino
default_toolchain: rpipicow
deployment_cookiecutter: https://github.com/fprime-community/fprime-arduino-deployment-cookiecutter.git
```

Some observations:

- I can build Ref in fprime, so I don't think it's my build system 
- I've tried updating the fprime build tools `pip install -U fprime-tools`

Any ideas? Let me know if you need any more information

LeStarch, 2024-04-01:
This builds on-top of `arduino-cli`. You should install that, and make sure you have the board packages installed.

LeStarch, 2024-04-01:
This repo will work you through the setup: https://github.com/fprime-community/fprime-baremetal-reference

csmith608, 2024-04-01:
@LeStarch I have the arduino-cli installed and working as well as the rp2040 board package.

LeStarch, 2024-04-01:
@csmith608 is arduino-cli on your PATH?

csmith608, 2024-04-01:
```
(fprime-venv) anniesmith@cs-mbp-2022 led-blinker % arduino-cli version
arduino-cli  Version: 0.35.3 Commit: 95cfd654 Date: 2024-02-19T13:24:24Z
(fprime-venv) anniesmith@cs-mbp-2022 led-blinker % ls /Users/anniesmith/led-blinker/fprime-arduino/cmake/toolchain 
ATmega128.cmake		featherrp2040rfm.cmake	support			teensy41.cmake
esp32.cmake		rpipico.cmake		teensy32.cmake		thingplusrp2040.cmake
featherM0.cmake		rpipicow.cmake		teensy40.cmake
```

LeStarch, 2024-04-01:
There is a bug in the arduino-cli-wrapper package `TypeError: 'type' object is not subscriptable`

LeStarch, 2024-04-01:
I'd recommend: 

1. Edit `/Users/your-user/fprime-venv/lib/python3.8/site-packages/arduino_cli_cmake_wrapper/cli.py` on line `193`  to capitalize the word `dict` from the text above
2. Run `purge` and `generate`.  

If this fixes the problem, then we can patch that script.  If more problems abound, run `fprime-util version-check` so we can try to reproduce.

csmith608, 2024-04-02:
The fixed that, then it complained about the TcpClient send port, so I changed it from 

```
@ Takes data to transmit out the UART device
        guarded input port send: Drv.ByteStreamSend
```

To
```
@ Takes data to transmit out the UART device
        guarded input port $send: Drv.ByteStreamSend
```

I don't think I can make the PR for the arduino-cli-cmake-wrapper, but I could for the TcpClient

ethancheez, 2024-04-21:
I fixed the `$send` port in TcpClient with [this commit](https://github.com/fprime-community/fprime-arduino/commit/42cfeb27ce041ec206798ff98f991407090bb4f6)

The fprime-arduino devel branch should build for the RP2040 now.

Gonta01, 2024-07-09:
fprime-util generate fails with logs:
[image]

fprime-util version-check:
[image]

arduino-cli version:
[image]

Does anyone knows how to fix it or what could be the reason?

Gonta01, 2024-07-10:
> Did you run:
> 
> arduino-cli lib install Time

@ethancheez thank you, it helped. Generate was successfully finished, but the next step (build) fails:
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ fprime-util build
[  0%] Built target version
[  0%] Building CXX object LedBlinker/CMakeFiles/fprime_arduino_patcher.dir/__/fprime-arduino/cmake/toolchain/support/sources/extras.cpp.obj
[  1%] Linking CXX static library ../lib/rpipicow/libfprime_arduino_patcher.a
[  1%] Built target fprime_arduino_patcher
[  1%] Generating ProcTypeEnumAi.xml
[  2%] Generating FppConstantsAc.cpp, FppConstantsAc.hpp, ProcTypeEnumAc.cpp, ProcTypeEnumAc.hpp
[  2%] Building CXX object config/CMakeFiles/config.dir/FppConstantsAc.cpp.obj
[  2%] Building CXX object config/CMakeFiles/config.dir/ProcTypeEnumAc.cpp.obj
[  2%] Linking CXX static library ../lib/rpipicow/libconfig.a
[  2%] Built target config
[  2%] Building CXX object F-Prime/Fw/Cfg/CMakeFiles/Fw_Cfg.dir/ConfigCheck.cpp.obj
[  2%] Linking CXX static library ../../../lib/rpipicow/libFw_Cfg.a
[  2%] Built target Fw_Cfg
[  2%] Generating ActiveEnumAi.xml, DeserialStatusEnumAi.xml, DirectionEnumAi.xml, EnabledEnumAi.xml, HealthEnumAi.xml, LogicEnumAi.xml, OnEnumAi.xml, OpenEnumAi.xml, SerialStatusEnumAi.xml, SuccessEnumAi.xml
[  2%] Generating ActiveEnumAc.cpp, ActiveEnumAc.hpp, DeserialStatusEnumAc.cpp, DeserialStatusEnumAc.hpp, DirectionEnumAc.cpp, DirectionEnumAc.hpp, EnabledEnumAc.cpp, EnabledEnumAc.hpp, HealthEnumAc.cpp, HealthEnumAc.hpp, LogicEnumAc.cpp, LogicEnumAc.hpp, OnEnumAc.cpp, OnEnumAc.hpp, OpenEnumAc.cpp, OpenEnumAc.hpp, SerialStatusEnumAc.cpp, SerialStatusEnumAc.hpp, SuccessEnumAc.cpp, SuccessEnumAc.hpp
[  2%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/Assert.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/String.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/InternalInterfaceString.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/MallocAllocator.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/MemAllocator.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/PolyType.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/SerialBuffer.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/Serializable.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/StringType.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/StringUtils.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/ActiveEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/DeserialStatusEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/DirectionEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/EnabledEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/HealthEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/LogicEnumAc.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/OnEnumAc.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/OpenEnumAc.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/SerialStatusEnumAc.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/SuccessEnumAc.cpp.obj
[  6%] Linking CXX static library ../../../lib/rpipicow/libFw_Types.a
[  6%] Built target Fw_Types
[  6%] Building CXX object F-Prime/Fw/Logger/CMakeFiles/Fw_Logger.dir/Logger.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Logger/CMakeFiles/Fw_Logger.dir/LogAssert.cpp.obj
[  6%] Linking CXX static library ../../../lib/rpipicow/libFw_Logger.a
[  6%] Built target Fw_Logger
[  7%] Building CXX object F-Prime/Fw/Obj/CMakeFiles/Fw_Obj.dir/ObjBase.cpp.obj
[  7%] Building CXX object F-Prime/Fw/Obj/CMakeFiles/Fw_Obj.dir/SimpleObjRegistry.cpp.obj
[  7%] Linking CXX static library ../../../lib/rpipicow/libFw_Obj.a
[  7%] Built target Fw_Obj
[  7%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/InputPortBase.cpp.obj
[  7%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/InputSerializePort.cpp.obj
[  7%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/OutputPortBase.cpp.obj
[  8%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/OutputSerializePort.cpp.obj
[  8%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/PortBase.cpp.obj
[  8%] Linking CXX static library ../../../lib/rpipicow/libFw_Port.a
[  8%] Built target Fw_Port
[  9%] Generating TimePortAi.xml
[  9%] Generating TimePortAc.cpp, TimePortAc.hpp
[  9%] Building CXX object F-Prime/Fw/Time/CMakeFiles/Fw_Time.dir/Time.cpp.obj
[  9%] Building CXX object F-Prime/Fw/Time/CMakeFiles/Fw_Time.dir/TimePortAc.cpp.obj
[  9%] Linking CXX static library ../../../lib/rpipicow/libFw_Time.a
[  9%] Built target Fw_Time
[ 10%] Generating ComPortAi.xml
[ 10%] Generating ComPortAc.cpp, ComPortAc.hpp
[ 10%] Building CXX object F-Prime/Fw/Com/CMakeFiles/Fw_Com.dir/ComPacket.cpp.obj
[ 10%] Building CXX object F-Prime/Fw/Com/CMakeFiles/Fw_Com.dir/ComBuffer.cpp.obj
[ 10%] Building CXX object F-Prime/Fw/Com/CMakeFiles/Fw_Com.dir/ComPortAc.cpp.obj
[ 11%] Linking CXX static library ../../../lib/rpipicow/libFw_Com.a
[... 95 lines cut ...]
      |            isAscii
[... cut ...]
  189 |         return (us * (clock_get_hz(clk_sys) / 1'000'000));
      |                ^
gmake[2]: *** [Arduino/Os/CMakeFiles/Arduino_Os.dir/build.make:244: Arduino/Os/CMakeFiles/Arduino_Os.dir/IntervalTimer.cpp.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:4852: Arduino/Os/CMakeFiles/Arduino_Os.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
[ERROR] CMake erred with return code 2

What am I missing now? And, by the way, there weren't in instruction this command "arduino-cli lib install Time" or I have missed something?

ethancheez, 2024-07-10:
> And, by the way, there weren't in instruction this command "arduino-cli lib install Time" or I have missed something?

The instructions for that are [here](https://github.com/fprime-community/fprime-arduino/blob/main/docs/arduino-cli-install.md#library-dependencies).

As for the build error, it is caused by `1'000'000` which is a syntax error in the RP2040 package itself when using the standard lower than `C++17` (I believe fprime uses `C++11` by default). I should have forced the compiler to use `C++17` standard [here](https://github.com/fprime-community/fprime-arduino/blob/main/cmake/toolchain/rpipicow.cmake#L6), but it looks like it may still be using an older standard. 

When you run `fprime-util generate rpipicow`, there should be something at the beginning that looks like this:
```
-- The C compiler identification is GNU 12.3.0
-- The CXX compiler identification is GNU 12.3.0
```
What version is yours reporting?

Gonta01, 2024-07-10:
> The instructions for that are [here](https://github.com/fprime-community/fprime-arduino/blob/main/docs/arduino-cli-install.md#library-dependencies).

Thank you. I have used [this one](https://fprime-community.github.io/fprime-tutorial-arduino-blinker/docs/arduino-cli-install.html) and it seems to be deprecated.

> What version is yours reporting?

```
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ fprime-util generate
[INFO] Generating build directory at: /home/user/Desktop/led-blinker/build-fprime-automatic-rpipicow
[INFO] Using toolchain file /home/user/Desktop/led-blinker/fprime-arduino/cmake/toolchain/rpipicow.cmake for platform default
-- The C compiler identification is GNU 12.3.0
-- The CXX compiler identification is GNU 12.3.0
```

Gonta01, 2024-07-10:
The suspicious is the following warning during build:
`cc1: warning: command-line option '-std=gnu++17' is valid for C++/ObjC++ but not for C`

ethancheez, 2024-07-10:
> What version is yours reporting?

Mine is also reporting GNU 12.3.0.

> The suspicious is the following warning during build: 
`cc1: warning: command-line option '-std=gnu++17' is valid for C++/ObjC++ but not for C`

I am also getting this, but mine successfully builds. I will try some tests to see if I can reproduce your error.

Gonta01, 2024-07-10:
> I will try some tests to see if I can reproduce your error.
Okay. I will be waiting for your feedback. I'm ready to give necessary information for you to reproduce the problem.

ethancheez, 2024-07-10:
Can you show me your output of `pip list | grep fprime`? And also which branches are you on for both `fprime` and `fprime-arduino`?

Gonta01, 2024-07-10:
```
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ pip list | grep fprime
fprime-fpl-convert-xml    1.0.3
fprime-fpl-extract-xml    1.0.3
fprime-fpl-layout         1.0.3
fprime-fpl-write-pic      1.0.3
fprime-fpp-check          2.1.0a3
fprime-fpp-depend         2.1.0a3
fprime-fpp-filenames      2.1.0a3
fprime-fpp-format         2.1.0a3
fprime-fpp-from-xml       2.1.0a3
fprime-fpp-locate-defs    2.1.0a3
fprime-fpp-locate-uses    2.1.0a3
fprime-fpp-syntax         2.1.0a3
fprime-fpp-to-cpp         2.1.0a3
fprime-fpp-to-json        2.1.0a3
fprime-fpp-to-xml         2.1.0a3
fprime-gds                3.4.3
fprime-tools              3.4.4
fprime-visual             1.0.2
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ cmake --version
cmake version 3.22.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ g++ --version
g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

```

```
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker/fprime-arduino$ git show-branch
* [devel] Fix $send escape for TcpClient
 ! [main] fprime v3.4.3 support
--
*  [devel] Fix $send escape for TcpClient
*  [devel^] Teensyduino 1.59 now uses C++17 standard
*  [devel~2] Ensure data is available in the stream before attempting to allocate/read
 - [main] fprime v3.4.3 support
-- [devel~3] bump to fprime v3.4.3
```

```
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker/fprime$ git show-branch
[devel] Rename EventsInterface to EventInterface (#2792)
```

ethancheez, 2024-07-10:
Try out fprime release v3.4.3. In the fprime directory, run `git checkout v3.4.3`

Gonta01, 2024-07-10:
It seems that amount of errors decreased:
```
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ fprime-util build
[  0%] Built target version
[  0%] Building CXX object LedBlinker/CMakeFiles/fprime_arduino_patcher.dir/__/fprime-arduino/cmake/toolchain/support/sources/extras.cpp.obj
[  1%] Linking CXX static library ../lib/rpipicow/libfprime_arduino_patcher.a
[  1%] Built target fprime_arduino_patcher
[  1%] Generating ProcTypeEnumAi.xml
[  2%] Generating FppConstantsAc.cpp, FppConstantsAc.hpp, ProcTypeEnumAc.cpp, ProcTypeEnumAc.hpp
[  2%] Building CXX object config/CMakeFiles/config.dir/FppConstantsAc.cpp.obj
[  2%] Building CXX object config/CMakeFiles/config.dir/ProcTypeEnumAc.cpp.obj
[  2%] Linking CXX static library ../lib/rpipicow/libconfig.a
[  2%] Built target config
[  2%] Building CXX object F-Prime/Fw/Cfg/CMakeFiles/Fw_Cfg.dir/ConfigCheck.cpp.obj
[  2%] Linking CXX static library ../../../lib/rpipicow/libFw_Cfg.a
[  2%] Built target Fw_Cfg
[  2%] Generating ActiveEnumAi.xml, DeserialStatusEnumAi.xml, DirectionEnumAi.xml, EnabledEnumAi.xml, HealthEnumAi.xml, LogicEnumAi.xml, OnEnumAi.xml, OpenEnumAi.xml, SerialStatusEnumAi.xml, SuccessEnumAi.xml
[  2%] Generating ActiveEnumAc.cpp, ActiveEnumAc.hpp, DeserialStatusEnumAc.cpp, DeserialStatusEnumAc.hpp, DirectionEnumAc.cpp, DirectionEnumAc.hpp, EnabledEnumAc.cpp, EnabledEnumAc.hpp, HealthEnumAc.cpp, HealthEnumAc.hpp, LogicEnumAc.cpp, LogicEnumAc.hpp, OnEnumAc.cpp, OnEnumAc.hpp, OpenEnumAc.cpp, OpenEnumAc.hpp, SerialStatusEnumAc.cpp, SerialStatusEnumAc.hpp, SuccessEnumAc.cpp, SuccessEnumAc.hpp
[  2%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/Assert.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/String.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/InternalInterfaceString.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/MallocAllocator.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/MemAllocator.cpp.obj
[  3%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/PolyType.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/SerialBuffer.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/Serializable.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/StringType.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/StringUtils.cpp.obj
[  4%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/ActiveEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/DeserialStatusEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/DirectionEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/EnabledEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/HealthEnumAc.cpp.obj
[  5%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/LogicEnumAc.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/OnEnumAc.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/OpenEnumAc.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/SerialStatusEnumAc.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Types/CMakeFiles/Fw_Types.dir/SuccessEnumAc.cpp.obj
[  6%] Linking CXX static library ../../../lib/rpipicow/libFw_Types.a
[  6%] Built target Fw_Types
[  6%] Building CXX object F-Prime/Fw/Logger/CMakeFiles/Fw_Logger.dir/Logger.cpp.obj
[  6%] Building CXX object F-Prime/Fw/Logger/CMakeFiles/Fw_Logger.dir/LogAssert.cpp.obj
[  6%] Linking CXX static library ../../../lib/rpipicow/libFw_Logger.a
[  6%] Built target Fw_Logger
[  7%] Building CXX object F-Prime/Fw/Obj/CMakeFiles/Fw_Obj.dir/ObjBase.cpp.obj
[  7%] Building CXX object F-Prime/Fw/Obj/CMakeFiles/Fw_Obj.dir/SimpleObjRegistry.cpp.obj
[  7%] Linking CXX static library ../../../lib/rpipicow/libFw_Obj.a
[  7%] Built target Fw_Obj
[  7%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/InputPortBase.cpp.obj
[  7%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/InputSerializePort.cpp.obj
[  7%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/OutputPortBase.cpp.obj
[  8%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/OutputSerializePort.cpp.obj
[  8%] Building CXX object F-Prime/Fw/Port/CMakeFiles/Fw_Port.dir/PortBase.cpp.obj
[  8%] Linking CXX static library ../../../lib/rpipicow/libFw_Port.a
[  8%] Built target Fw_Port
[  9%] Generating TimePortAi.xml
[  9%] Generating TimePortAc.cpp, TimePortAc.hpp
[  9%] Building CXX object F-Prime/Fw/Time/CMakeFiles/Fw_Time.dir/Time.cpp.obj
[  9%] Building CXX object F-Prime/Fw/Time/CMakeFiles/Fw_Time.dir/TimePortAc.cpp.obj
[  9%] Linking CXX static library ../../../lib/rpipicow/libFw_Time.a
[  9%] Built target Fw_Time
[ 10%] Generating ComPortAi.xml
[ 10%] Generating ComPortAc.cpp, ComPortAc.hpp
[ 10%] Building CXX object F-Prime/Fw/Com/CMakeFiles/Fw_Com.dir/ComPacket.cpp.obj
[ 10%] Building CXX object F-Prime/Fw/Com/CMakeFiles/Fw_Com.dir/ComBuffer.cpp.obj
[ 10%] Building CXX object F-Prime/Fw/Com/CMakeFiles/Fw_Com.dir/ComPortAc.cpp.obj
[ 11%] Linking CXX static library ../../../lib/rpipicow/libFw_Com.a
[ 11%] Built target Fw_Com
[ 12%] Generating TlmGetPortAi.xml, TlmPortAi.xml
[ 12%] Generating TlmGetPortAc.cpp, TlmGetPortAc.hpp, TlmPortAc.cpp, TlmPortAc.hpp
[... 74 lines cut ...]
[... cut ...]
/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.3/cores/rp2040/api/../../../ArduinoCore-API/api/WCharacter.h:148:10: error: 'toascii' was not declared in this scope; did you mean 'toAscii'?
  148 |   return toascii (c);
      |          ^~~~~~~
      |          toAscii
gmake[2]: *** [Arduino/Os/CMakeFiles/Arduino_Os.dir/build.make:244: Arduino/Os/CMakeFiles/Arduino_Os.dir/IntervalTimer.cpp.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:4852: Arduino/Os/CMakeFiles/Arduino_Os.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
[ERROR] CMake erred with return code 2
```

Gonta01, 2024-07-10:
It seems to me that errors related to "1'000'000" disappeared after I modified settings.cmake

Gonta01, 2024-07-10:
I have made brute force fix of the problem by replacing  isascii (c) with (((c) & ~0x7f) == 0) toascii(c) with ((c) & 0x7f).
I have decided to do it based on discussion [WCharacter.h - isascii not declared in this scope on arm-none-eabi ](https://github.com/arduino/ArduinoCore-API/issues/158). The discussion ended with the following [fix ](https://github.com/arduino/ArduinoCore-API/commit/1171abba2c3170c8a93a08f45119594805e13446)

Gonta01, 2024-07-10:
I've made changes in WCharacter.h file:
[image]
[image]

Gonta01, 2024-07-10:
@ethancheez what is you fprime-util version-check and arduino-cli version?

> fprime-util version-check:
> 
> ```
> (fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ fprime-util version-check
> Operating System: Linux
> CPU Architecture: x86_64
> Platform: Linux-6.5.0-41-generic-x86_64-with-glibc2.35
> Python version: 3.10.12
> CMake version: 3.22.1
> Pip version: 24.1.2
> Pip packages:
>     fprime-tools==3.4.4
>     fprime-gds==3.4.3
>     fprime-fpp-*==2.1.0a3
> ```
> 
> arduino-cli version:
> 
> ```
> (fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ arduino-cli version
> arduino-cli  Version: 1.0.2 Commit: 33dfa8e8 Date: 2024-07-02T12:53:01Z
> ```

ethancheez, 2024-07-10:
> @ethancheez what is you fprime-util version-check and arduino-cli version?

```sh
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-4.4.0-22621-Microsoft-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.29.6
Pip version: 24.1.2
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3
```
```sh
arduino-cli  Version: 1.0.2 Commit: 33dfa8e8 Date: 2024-07-02T12:53:01Z
```

Let's try some things to avoid having to edit the RP2040 package files and fprime core files. In `fprime-arduino/cmake/toolchain/rpipicow.cmake` line 17, change it to:
```cmake
add_compile_options(-D_BOARD_RPIPICOW -DUSE_BASIC_TIMER -D_GNU_SOURCE)
```

ethancheez, 2024-07-11:
Can you try changing to this in the `rpipicow.cmake` file:
```cmake
add_compile_options(-std=c++17 -std=gnu++17 -std=gnu17) # Force C++17 standard
```
This will add a bunch of warnings saying something like `command-line option '-std=gnu17' is valid for C/ObjC but not for C++`, you can ignore it since the `-std=gnu++17` should handle C++

Gonta01, 2024-07-12:
@ethancheez, neither the first approach, nor the second one helped. By the way, was you able to reproduce this issue on your side? Is this only on my side?

ethancheez, 2024-07-12:
> @ethancheez, neither the first approach, nor the second one helped. By the way, was you able to reproduce this issue on your side? Is this only on my side?

I was not able to reproduce it. I tried a completely new Ubuntu VM and reinstalled everything and successfully builds. Do you mind listing the steps you took to create your project?

Gonta01, 2024-07-23:
> > @ethancheez, neither the first approach, nor the second one helped. By the way, was you able to reproduce this issue on your side? Is this only on my side?
> 
> I was not able to reproduce it. I tried a completely new Ubuntu VM and reinstalled everything and successfully builds. Do you mind listing the steps you took to create your project?
> 
> These were my steps (the important parts). If you followed the tutorial step by step, you would run `fprime-bootstrap project` instead of `fprime-bootstrap clone`
> 
> ```shell
> python -m venv ~/venv/fprime
> . ~/venv/fprime/bin/activate
> pip install fprime-bootstrap
> fprime-bootstrap clone https://github.com/fprime-community/fprime-tutorial-arduino-blinker.git
> cd fprime-tutorial-arduino-blinker
> . ~/fprime-venv/bin/activate
> 
> curl -fsSL https://raw.githubusercontent.com/arduino/arduino-cli/master/install.sh | BINDIR=$VIRTUAL_ENV/bin sh
> pip install arduino-cli-cmake-wrapper
> 
> arduino-cli config init
> arduino-cli config add board_manager.additional_urls https://www.pjrc.com/teensy/package_teensy_index.json
> arduino-cli config add board_manager.additional_urls https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
> arduino-cli config add board_manager.additional_urls https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
> arduino-cli config add board_manager.additional_urls https://mcudude.github.io/MegaCore/package_MCUdude_MegaCore_index.json
> arduino-cli config add board_manager.additional_urls https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json
> 
> arduino-cli core update-index
> arduino-cli core install teensy:avr
> arduino-cli core install adafruit:samd
> arduino-cli core install esp32:esp32
> arduino-cli core install MegaCore:avr
> arduino-cli core install rp2040:rp2040
> 
> arduino-cli lib install Time
> 
> fprime-util generate rpipicow
> fprime-util build rpipicow -j8
> ```

Followed through your steps and faced with the same problem:

```
(fprime-venv) user@user-VirtualBox:~/Desktop/fprime-tutorial-arduino-blinker$ fprime-util generate rpipicow

[INFO] Generating build directory at: /home/user/Desktop/fprime-tutorial-arduino-blinker/build-fprime-automatic-rpipicow

[INFO] Using toolchain file /home/user/Desktop/fprime-tutorial-arduino-blinker/fprime-arduino/cmake/toolchain/rpipicow.cmake for platform rpipicow

-- The C compiler identification is GNU 12.3.0

-- The CXX compiler identification is GNU 12.3.0

-- Detecting C compiler ABI info

-- Detecting C compiler ABI info - done

-- Check for working C compiler: /home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/arm-none-eabi-gcc - skipped

-- Detecting C compile features

-- Detecting C compile features - done

-- Detecting CXX compiler ABI info

-- Detecting CXX compiler ABI info - done

-- Check for working CXX compiler: /home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/arm-none-eabi-g++ - skipped

-- Detecting CXX compile features

-- Detecting CXX compile features - done

[... 646 lines cut ...]
```
(fprime) user@user-VirtualBox:~/Desktop/fprime-tutorial-arduino-blinker$ . ~/fprime-venv/bin/activate
bash: /home/user/fprime-venv/bin/activate: No such file or directory
(fprime) user@user-VirtualBox:~/Desktop/fprime-tutorial-arduino-blinker$ . fprime-venv/bin/activate
```
```
(fprime-venv) user@user-VirtualBox:~/Desktop/fprime-tutorial-arduino-blinker$ arduino-cli config init
Config file already exists, use --overwrite to discard the existing one.
(fprime-venv) user@user-VirtualBox:~/Desktop/fprime-tutorial-arduino-blinker$ arduino-cli config init --overwrite
Config file written to: /home/user/.arduino15/arduino-cli.yaml
```

Here is my version of Python and Ubuntu:
```
(fprime-venv) user@user-VirtualBox:~/Desktop/fprime-tutorial-arduino-blinker$ python -V
Python 3.10.12
(fprime-venv) user@user-VirtualBox:~/Desktop/fprime-tutorial-arduino-blinker$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
```

@ethancheez what is your version of Ubuntu and Python?

ethancheez, 2024-07-23:
> @ethancheez what is your version of Ubuntu and Python?

```sh
ethan@ethancheez:~$ python --version
Python 3.10.12
ethan@ethancheez:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy
```

Gonta01, 2024-07-25:
@ethancheez hi! Could you please add the following lines to your rpipicow.cmake:
```
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -v")
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS} -v")
```
[image]

Show me your logs for build of some object (let it be Fw_Logger.dir/Logger.cpp.obj) :
```
[  8%] Building CXX object F-Prime/Fw/Logger/CMakeFiles/Fw_Logger.dir/Logger.cpp.obj
Using built-in specs.
COLLECT_GCC=/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/arm-none-eabi-g++
Target: arm-none-eabi
Configured with: /workdir/repo/gcc-gnu/configure --prefix=/workdir/arena.x86_64/arm-none-eabi --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=arm-none-eabi --disable-shared --with-newlib --enable-threads=no --disable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-nls --without-python --disable-bootstrap --enable-languages=c,c++ --disable-lto --enable-static=yes --disable-libstdcxx-verbose --disable-decimal-float --with-cpu=cortex-m0plus --with-no-thumb-interwork --disable-tui --disable-pie-tools --disable-libquadmath
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.3.0 (GCC) 
COLLECT_GCC_OPTIONS='-D' 'FW_BAREMETAL_SCHEDULER=1' '-I' '/home/user/Desktop/led-blinker/build-fprime-automatic-rpipicow/arduino-cli-sketch/core' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/include' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/cores/rp2040' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/variants/rpipicow' '-I' '/home/user/Desktop/led-blinker/fprime-arduino/cmake/platform/types/arm' '-I' '/home/user/Desktop/led-blinker/fprime-arduino/cmake/platform' '-I' '/home/user/Desktop/led-blinker/fprime' '-I' '/home/user/Desktop/led-blinker/fprime/config' '-I' '/home/user/Desktop/led-blinker' '-I' '/home/user/Desktop/led-blinker/build-fprime-automatic-rpipicow' '-I' '/home/user/Desktop/led-blinker/build-fprime-automatic-rpipicow/F-Prime' '-I' '/home/user/Desktop/led-blinker/build-fprime-automatic-rpipicow/config' '-I' '/home/user/Desktop/led-blinker/fprime-arduino' '-I' '/home/user/Arduino/libraries/Time' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/Wire/src' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/SPI/src' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/WiFi/src' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/lwIP_CYW43/src' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/lwIP_Ethernet/src' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/Updater/src' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/MD5Builder/src' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/LittleFS/src' '-I' '/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/PicoOTA/src' '-v' '-v' '-std=c++17' '-std=gnu++17' '-D' '_BOARD_RPIPICOW' '-D' 'USE_BASIC_TIMER' '-D' '_GNU_SOURCE' '-std=c++11' '-D' 'ASSERT_FILE_ID=0x2c35dd08' '-D' 'ASSERT_RELATIVE_PATH="Fw/Logger/Logger.cpp"' '-MD' '-MT' 'F-Prime/Fw/Logger/CMakeFiles/Fw_Logger.dir/Logger.cpp.obj' '-MF' 'CMakeFiles/Fw_Logger.dir/Logger.cpp.obj.d' '-o' 'CMakeFiles/Fw_Logger.dir/Logger.cpp.obj' '-c' '-mcpu=cortex-m0plus' '-mthumb' '-mlibarch=armv6s-m' '-march=armv6s-m' '-dumpdir' 'CMakeFiles/Fw_Logger.dir/'
 /home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../libexec/gcc/arm-none-eabi/12.3.0/cc1plus -quiet -v -v -I /home/user/Desktop/led-blinker/build-fprime-automatic-rpipicow/arduino-cli-sketch/core -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/include -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/cores/rp2040 -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/variants/rpipicow -I /home/user/Desktop/led-blinker/fprime-arduino/cmake/platform/types/arm -I /home/user/Desktop/led-blinker/fprime-arduino/cmake/platform -I /home/user/Desktop/led-blinker/fprime -I /home/user/Desktop/led-blinker/fprime/config -I /home/user/Desktop/led-blinker -I /home/user/Desktop/led-blinker/build-fprime-automatic-rpipicow -I /home/user/Desktop/led-blinker/build-fprime-automatic-rpipicow/F-Prime -I /home/user/Desktop/led-blinker/build-fprime-automatic-rpipicow/config -I /home/user/Desktop/led-blinker/fprime-arduino -I /home/user/Arduino/libraries/Time -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/Wire/src -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/SPI/src -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/WiFi/src -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/lwIP_CYW43/src -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/lwIP_Ethernet/src -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/Updater/src -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/MD5Builder/src -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/LittleFS/src -I /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/libraries/PicoOTA/src -imultilib thumb -iprefix /home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/ -MD CMakeFiles/Fw_Logger.dir/Logger.cpp.d -MF CMakeFiles/Fw_Logger.dir/Logger.cpp.obj.d -MT F-Prime/Fw/Logger/CMakeFiles/Fw_Logger.dir/Logger.cpp.obj -D__USES_INITFINI__ -D FW_BAREME [... cut ...] in/../lib/gcc/arm-none-eabi/12.3.0/:/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/:/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/lib/
```

Suspicious moments:
`ignoring nonexistent directory "/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/sys-include"`

`GNU C++11 (GCC) version 12.3.0 (arm-none-eabi)`

ethancheez, 2024-07-25:
```
[ 13%] Building CXX object F-Prime/Utils/Hash/CMakeFiles/Utils_Hash.dir/HashCommon.cpp.obj
GNU C++17 (GCC) version 12.3.0 (arm-none-eabi)
        compiled by GNU C version 9.4.0, GMP version 6.1.2, MPFR version 3.1.4, MPC version 1.0.3, isl version isl-0.18-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Using built-in specs.
ignoring nonexistent directory "/home/ethan/.arduino15/packages/rp2040/hardware/rp2040/3.9.3/../../../../arm-none-eabi/include/c++/12.3.0"
COLLECT_GCC=/home/ethan/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/arm-none-eabi-g++
ignoring nonexistent directory "/home/ethan/.arduino15/packages/rp2040/hardware/rp2040/3.9.3/../../../../arm-none-eabi/include/c++/12.3.0/arm-none-eabi/thumb"
Target: arm-none-eabi
cc1: warning: command-line option '-std=c++17' is valid for C++/ObjC++ but not for C
ignoring nonexistent directory "/home/ethan/.arduino15/packages/rp2040/hardware/rp2040/3.9.3/../../../../arm-none-eabi/include/c++/12.3.0/backward"
Configured with: /workdir/repo/gcc-gnu/configure --prefix=/workdir/arena.x86_64/arm-none-eabi --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=arm-none-eabi --disable-shared --with-newlib --enable-threads=no --disable-__cxa_atexit --disable-libgomp --disable-libmudflap --disable-nls --without-python --disable-bootstrap --enable-languages=c,c++ --disable-lto --enable-static=yes --disable-libstdcxx-verbose --disable-decimal-float --with-cpu=cortex-m0plus --with-no-thumb-interwork --disable-tui --disable-pie-tools --disable-libquadmath
cc1: warning: command-line option '-std=gnu++17' is valid for C++/ObjC++ but not for C
ignoring nonexistent directory "/home/ethan/.arduino15/packages/rp2040/hardware/rp2040/3.9.3/include-fixed"
Thread model: single
ignoring nonexistent directory "/home/ethan/.arduino15/packages/rp2040/hardware/rp2040/3.9.3/../../../../arm-none-eabi/sys-include"
Supported LTO compression algorithms: zlibignoring nonexistent directory "/home/ethan/.arduino15/packages/rp2040/hardware/rp2040/3.9.3/../../../../arm-none-eabi/include"

ignoring nonexistent directory "/home/ethan/.arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/../../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/sys-include"
gcc version 12.3.0 (GCC) 
```
Looks like my build is using C++17.

I also get the `ignoring nonexxistent directory` messages, but successfully builds.

Gonta01, 2024-07-26:
@ethancheez have you any ideas, in what directions I should watch to debug this issue.

Gonta01, 2024-07-26:
I've followed through [steps ](https://github.com/fprime-community/fprime-arduino/issues/27#issuecomment-2224862122)that you described on WSL1 with Ubuntu22.04 but result again the same.

ethancheez, 2024-07-27:
Something I noticed in your verbose output, take a look at the `COLLECT_GCC_OPTIONS=` output. Near the end, I see:

```sh
 `'-v' '-v' '-std=c++17' '-std=gnu++17' '-D' '_BOARD_RPIPICOW' '-D' 'USE_BASIC_TIMER' '-D' '_GNU_SOURCE' '-std=c++11`
```
The `-std=c++11` appears last, and therefore takes precedence over the `-std=c++17` flag. 

Here is my order:
```sh
'-v' '-std=c++11' '-std=c++17' '-std=gnu++17' '-D' '_BOARD_RPIPICOW' '-D' 'USE_BASIC_TIMER'
```

Now we just have to figure out why the `-std=c++11` flag was added last, as all the flags added via `add_compile_options` within the `rpipicow.cmake` file should be last.

Some random things to try:
- Have you been running `fprime-util build rpipicow` with a `-j` flag? Maybe try run it with `-j1` to force only one job (although I don't think this will do anything).
- Try also update your `cmake` version. My Ubuntu VM uses `3.28.3` and my WSL1 uses `3.29.6`, both works.
- Try move the `add_compile_options(-std=c++17 -std=gnu++17) # Force C++17 standard` to the very end of the `rpipicow.cmake` file, right after the `include("${CMAKE_CURRENT_LIST_DIR}/support/arduino-support.cmake")`.
- Try build for `teensy41` by running: `fprime-util generate teensy41` and `fprime-util build teensy41` with the verbose flags turned on and see the order of the `-std=c++17` and `-std=c++11` being printed.

Gonta01, 2024-07-29:
@ethancheez update of Cmake helped! Everything seems to build without any issues now.
I updated cmake from 3.22.1 to 3.30.1:

```
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ cmake -version
cmake version 3.22.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ sudo apt remove cmake -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  cmake-data dh-elpa-helper libjsoncpp25 librhash0 libwpe-1.0-1 libwpebackend-fdo-1.0-1
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  cmake
0 upgraded, 0 newly installed, 1 to remove and 8 not upgraded.
After this operation, 21,2 MB disk space will be freed.
(Reading database ... 193709 files and directories currently installed.)
Removing cmake (3.22.1-1ubuntu1.22.04.2) ...
Processing triggers for man-db (2.10.2-1) ...
(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ pip install cmake --upgrade
Collecting cmake
  Downloading cmake-3.30.1-py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (6.1 kB)
Downloading cmake-3.30.1-py3-none-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (26.9 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 26.9/26.9 MB 18.6 MB/s eta 0:00:00
Installing collected packages: cmake
Successfully installed cmake-3.30.1

[notice] A new release of pip is available: 24.1.2 -> 24.2
[notice] To update, run: pip install --upgrade pip

(fprime-venv) user@user-VirtualBox:~/Desktop/led-blinker$ cmake -version
cmake version 3.30.1

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

The order of flags now:
`'-std=c++11' '-std=c++17' '-std=gnu++17' '-D' '_BOARD_RPIPICOW' '-D' 'USE_BASIC_TIMER' '-D'`

@ethancheez thank you a lot for you help and work!

Gonta01, 2024-07-29:
Just to clarify does your project with activated verbose flags fully build?
When I activate them the build (rpipicow) fails at the stage: 

`[ 26%] Building CXX object Arduino/Os/CMakeFiles/Arduino_Os.dir/IntervalTimer.cpp.obj
`

```
In file included from /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/cores/rp2040/api/../../../ArduinoCore-API/api/ArduinoAPI.h:30,
                 from /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/cores/rp2040/api/ArduinoAPI.h:2,
                 from /home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/cores/rp2040/Arduino.h:28,
                 from /home/user/Desktop/led-blinker/fprime-arduino/cmake/platform/FprimeArduino.hpp:9,
                 from /home/user/Desktop/led-blinker/fprime-arduino/Arduino/Os/IntervalTimer.cpp:7:
/home/user/.arduino15/packages/rp2040/hardware/rp2040/3.9.4/cores/rp2040/api/../../../ArduinoCore-API/api/IPAddress.h:26:10: fatal error: lwip/init.h: No such file or directory
   26 | #include <lwip/init.h>
      |          ^~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [Arduino/Os/CMakeFiles/Arduino_Os.dir/build.make:244: Arduino/Os/CMakeFiles/Arduino_Os.dir/IntervalTimer.cpp.obj] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:4867: Arduino/Os/CMakeFiles/Arduino_Os.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
[ERROR] CMake erred with return code 2
```
Without verbose flags everything seems to work fine as I described above.

ethancheez, 2024-07-29:
> Just to clarify does your project with activated verbose flags fully build? When I activate them the build (rpipicow) fails at the stage:
>
> `[ 26%] Building CXX object [...]

Yeah I have been getting the same error with the verbose flag. Unsure why. Seems to only happen when you run `fprime-util generate rpipicow` and `fprime-util build rpipicow` for the very first time with the verbose flags present. If you build first without the verbose flags, let it successfully build, then add it into the `rpipicow.cmake` after, and then rebuild, it will succeed with all the verbose outputs.

Gonta01, 2024-07-29:
Got it. Thank you.
I'm curious why that problem with cmake even happened. I installed all [System Requirements](https://nasa.github.io/fprime/INSTALL.html) step by step and for some reason had deprecated version. Is it possible to add system requirements (despite python, pip, venv) to requirements.txt, so that they will be part of virtual environment? Are there any other solution to prevent such kind of problems in the future?

ethancheez, 2024-08-01:
> Got it. Thank you. I'm curious why that problem with cmake even happened. I installed all [System Requirements](https://nasa.github.io/fprime/INSTALL.html) step by step and for some reason had [...]

I believe the system requirements for F Prime itself is CMake 3.16 or newer, which you did have. It's just the RPi Pico Arduino package requires a newer CMake version. It might be possible to add it to the `requirements.txt` file within the arduino blinker tutorial.

LeStarch, 2024-08-01:
This could be added as a `cmake_required_version` call in the rp2040 cmake file.

Gonta01, 2024-08-02:
@ethancheez and @LeStarch thank you for your responses and suggested solutions. 

` It's just the RPi Pico Arduino package requires a newer CMake version.` - this problem will occur in those people who'll try to do their own cross-compilation . The toolchain cmake doesn't take precedence over settings.cmake of fprime.

ethancheez, 2024-11-01:
Closing. Issue resolved, will add the `cmake_minimum_required` flag in the cmake files in future branch.

closed 2024-11-01 by ethancheez