fprime-community/fprime-arduino #31: fprime v3.5.0 compatibility issues

nateinaction, 2024-10-24:
Using the rp2040 as a target I notice that the 3.4->3.5 upgrade fails with some compatibility issues.

It looks like the baremetal os files were removed from `fprime` during the OSAL and task refactors leading to the v3.5.0 release. 

Related PRs:
Task refactor https://github.com/nasa/fprime/pull/2672
OSAL cleanup https://github.com/nasa/fprime/pull/2933

Error:
```
-- Configuring done (51.7s)
CMake Error at fprime/cmake/target/build.cmake:55 (set_target_properties):
  Cannot find source file:

    /Users/nate/code/github.com/proveskit/fprime-proves/fprime/Os/TaskCommon.cpp

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm
  .ccm .cxxm .c++m .h .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90
  .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  fprime/cmake/target/build.cmake:151 (build_setup_build_module)
  fprime/cmake/target/target.cmake:116 (build_add_module_target)
  fprime/cmake/target/target.cmake:116 (cmake_language)
  fprime/cmake/target/target.cmake:155 (setup_single_target)
  fprime/cmake/module.cmake:42 (setup_module_targets)
  fprime/cmake/module.cmake:82 (generate_base_module_properties)
  fprime/cmake/API.cmake:239 (generate_library)
  fprime-arduino/Arduino/Os/CMakeLists.txt:47 (register_fprime_module)

CMake Error at fprime/cmake/module.cmake:33 (add_library):
  No SOURCES given to target: Arduino_Os
Call Stack (most recent call first):
  fprime/cmake/module.cmake:82 (generate_base_module_properties)
  fprime/cmake/API.cmake:239 (generate_library)
  fprime-arduino/Arduino/Os/CMakeLists.txt:47 (register_fprime_module)
```

LeStarch, 2024-10-25:
Yes, this is being prepped as we speak!

ethancheez, 2024-10-26:
Try out the [`fprime-v3.5.0`](https://github.com/fprime-community/fprime-arduino/tree/fprime-v3.5.0) branch. Some of the OSAL implementations are still WIP, but I got a deployment (LedBlinker) running.

nateinaction, 2024-10-27:
@ethancheez Using commit 293b221 I'm seeing an error on `fprime-util generate`

### Environment
```
Operating System: Darwin
CPU Architecture: arm64
Platform: macOS-15.0.1-arm64-arm-64bit
Python version: 3.9.6
CMake version: 3.30.5
Pip version: 21.2.4
Pip packages:
    fprime-tools==3.5.0
    fprime-gds==3.5.0
    fprime-fpp-*==2.2.0
```

### Error
[image]

I have verified the board definition is installed:
```
arduino-cli board listall |grep rpipico
Raspberry Pi Pico                                                       rp2040:rp2040:rpipico
```

and I'm using a relatively recent version of `arduino-cli`
```
arduino-cli version
arduino-cli  Version: 1.0.4 Commit: Homebrew Date: 2024-08-12T13:32:50Z
```

I didn't see any changes to the `cmake/toolchain/support/arduino-wrapper.py` in this PR so I'm not sure why I'm running into this issue here and not in my v3.4.3 branch.

My working branch uses fprime v3.4.3 along with fprime-arduino ffd162f and rp2040 board definition installed with `arduino-cli core install rp2040:rp2040@3.9.5`.

I noticed that there seems to be [some work-in-progress](https://github.com/SterlingPeet/arduino-cli-cmake-wrapper/pull/10) in the arduino-cli cmake wrapper repo. Are you building with a different version of the wrapper than what I get when I `pip install arduino-cli-cmake-wrapper`?

nateinaction, 2024-10-27:
@ethancheez 

I'm running into an issue on `fprime-util generate` with lots of errors that look like

```
CMake Error at fprime/cmake/implementation.cmake:105 (target_sources):
  Error evaluating generator expression:

    $<TARGET_PROPERTY:Os_Cpu_Baremetal,REQUIRED_SOURCE_FILES>

  Target "Os_Cpu_Baremetal" not found.
Call Stack (most recent call first):
  fprime/cmake/implementation.cmake:64 (setup_executable_implementation)
  fprime/cmake/target/build.cmake:107 (setup_executable_implementations)
  fprime/cmake/target/build.cmake:151 (build_setup_build_module)
  fprime/cmake/target/build.cmake:126 (build_add_module_target)
  fprime/cmake/target/target.cmake:124 (build_add_deployment_target)
  fprime/cmake/target/target.cmake:124 (cmake_language)
  fprime/cmake/target/target.cmake:155 (setup_single_target)
  fprime/cmake/module.cmake:42 (setup_module_targets)
  fprime/cmake/module.cmake:55 (generate_base_module_properties)
  fprime/cmake/API.cmake:397 (generate_deployment)
  BroncoDeployment/CMakeLists.txt:27 (register_fprime_deployment)
```

I thought this might be something I need to update in my deployment so I copied over the [changes to the cookie cutter](https://github.com/fprime-community/fprime-arduino-deployment-cookiecutter/compare/main...fprime-v3.5.0) but that did not seem to help.

My deployment's `CMakeLists.txt` looks roughly identical to the cookie cutter:
```cmake
#####
# 'BroncoDeployment' Deployment:
#
# This registers the 'BroncoDeployment' deployment to the build system. 
# Custom components that have not been added at the project-level should be added to 
# the list below.
#
#####

# Deployment is only compatible with the Arduino platform
restrict_platforms(ArduinoFw)

###
# Topology and Components
###
add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/Top/")

# Add custom components to this specific deployment here
# add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/MyComponent/")

set(SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Main.cpp")
set(MOD_DEPS ${FPRIME_CURRENT_MODULE}/Top)

# choose_fprime_implementation("Os_File" Os_File_Stub)

register_fprime_deployment()

finalize_arduino_executable("${FPRIME_CURRENT_MODULE}")
```

ethancheez, 2024-10-27:
@nathancheek 
`fprime-arduino` now depends on [`fprime-baremetal`](https://github.com/fprime-community/fprime-baremetal/tree/update/v3.5.0-ethan) (branch `v3.5.0-ethan`). Do you have this as a submodule in your project?

And to make sure your project is importing `fprime-baremetal`, your `settings.ini` should have:
```
library_locations: ./fprime-arduino:./fprime-baremetal
```

nateinaction, 2024-10-29:
I'm getting closer. I'm able to generate without errors now. Then I ran into the fatal logger issue on build that you fixed in nasa/fprime#2986. Finally I upgraded my `fprime` submodule to the `devel` commit 0e4b4c2 (after your PR merged) and ran into the following error. Purging and attempting to regenerate results in the same error:
```
CMake Error at fprime/cmake/utilities.cmake:586 (message):
-- Configuring incomplete, errors occurred!
  [sub-build] Failed to generate: info-cache:

  CMake Error at
  fprime-arduino/cmake/toolchain/support/arduino-support.cmake:121
  (prevent_prescan):

    Unknown CMake command "prevent_prescan".

  Call Stack (most recent call first):

    fprime-arduino/cmake/toolchain/support/arduino-support.cmake:185 (setup_arduino_libraries)
    BroncoDeployment/CMakeLists.txt:29 (finalize_arduino_executable)
Call Stack (most recent call first):
  fprime/cmake/sub-build/sub-build.cmake:39 (execute_process_or_fail)
  fprime/cmake/FPrime.cmake:162 (run_sub_build)
  fprime/cmake/FPrime.cmake:231 (fprime_initialize_build_system)
  CMakeLists.txt:13 (include)
```

Looks like this macro was removed in nasa/fprime#2956. So I reverted `fprime` back to `tags/v3.5.0` and cherry-picked 0e4b4c2. This seemed to work and I was able to get to 99% of a build before finally failing with some kind of memory overrun:
```
[ 99%] Linking CXX executable ../bin/rpipico/BroncoDeployment
/Users/nate/Library/Arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: ../bin/rpipico/BroncoDeployment section `.bss' will not fit in region `RAM'
/Users/nate/Library/Arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: region RAM overflowed
/Users/nate/Library/Arduino15/packages/rp2040/tools/pqt-gcc/2.2.0-d04e724/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 342812 bytes
collect2: error: ld returned 1 exit status
gmake[2]: *** [BroncoDeployment/CMakeFiles/BroncoDeployment.dir/build.make:372: bin/rpipico/BroncoDeployment] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:7433: BroncoDeployment/CMakeFiles/BroncoDeployment.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
```

I tried upgrading my rp2040 board definition from 3.9.5 -> 4.1.1 and found a slightly different error in the same vein:
```
[100%] Linking CXX executable ../bin/rpipico/BroncoDeployment
<command-line>: note: this is the location of the previous definition
/Users/nate/Library/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld:/Users/nate/code/github.com/proveskit/fprime-proves-2/build-fprime-automatic-rpipico/arduino-cli-sketch/memmap_default.ld:235 cannot move location counter backwards (from 20094018 to 20040000)
collect2: error: ld returned 1 exit status
gmake[2]: *** [BroncoDeployment/CMakeFiles/BroncoDeployment.dir/build.make:373: bin/rpipico/BroncoDeployment] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:7433: BroncoDeployment/CMakeFiles/BroncoDeployment.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
[ERROR] CMake erred with return code 2
```

I'm not really sure where to go from here. Any thoughts on this?

ethancheez, 2024-10-29:
> Looks like this macro was removed in [nasa/fprime#2956](https://github.com/nasa/fprime/pull/2956). So I reverted `fprime` back to `tags/v3.5.0` and cherry-picked 0e4b4c2. This seemed to work and I [...]

A RAM overflow is very common in baremetal systems because there is much smaller onboard memory compared to beefier computers (like a raspberry pi). The RPi Pico has 264 KB RAM so it looks like your specific project exceeded that amount. The solution for this is to modify the `config/` files and play around with the constants to decrease the number of DB entries, telemetry buckets, etc. You can view an example `config/` setup [here](https://github.com/fprime-community/fprime-tutorial-arduino-blinker/tree/fprime-v3.5.0/config).

> I tried upgrading my rp2040 board definition from 3.9.5 -> 4.1.1 and found a slightly different error in the same vein:
>
> ```
> [100%] Linking CXX executable ../bin/rpipico/BroncoDeployment [...]

This is my first time seeing this error but I *believe* that it is equivalent to the prior error on `3.9.5`. This error is basically saying you have a memory overlap where data is being written into memory that has already been written into before.

Also, as a side note, I was unable to get any RP2040 toolchains building for versions newer than `3.9.5`. It looks like the RP2040 devs changed something starting from version `4.0.0` to support RP2350 boards. A `.bin` gets generated, but my Pico was unable to understand it after converting it to `.uf2`. The generated `.uf2` file is empty.

csmith608, 2024-11-03:
> A RAM overflow is very common in baremetal systems because there is much smaller onboard memory compared to beefier computers (like a raspberry pi). The RPi Pico has 264 KB RAM so it looks like your [...]

I'm having a similar issue with fprime 3.5

```
/Users/anniesmith/Library/Arduino15/packages/rp2040/tools/pqt-gcc/4.0.1-8ec9d6f/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld:/Users/anniesmith/LedBlinkerArduino/build-fprime-automatic-rpipicow/arduino-cli-sketch/memmap_default.ld:235 cannot move location counter backwards (from 2009fb10 to 20040000)
collect2: error: ld returned 1 exit status
```

ethancheez, 2024-11-03:
> I'm having a similar issue with fprime 3.5
>
> ``` [...]

Do you have a custom `config/` directory in your project with custom static sizes? The error looks like a memory overflow because your binary is too large.

csmith608, 2024-11-03:
> Do you have a custom `config/` directory in your project with custom static sizes? The error looks like a memory overflow because your binary is too large.

I don't, I could try that, but I get a similar error when compiling for the teensy 

```
/Users/anniesmith/Library/Arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: address 0x20094f20 of ../bin/teensy41/LedBlinker.elf section `.bss' is not within region `DTCM'
/Users/anniesmith/Library/Arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: address 0x20094f20 of ../bin/teensy41/LedBlinker.elf section `.bss' is not within region `DTCM'
collect2: error: ld returned 1 exit status
```

I wouldn't expect that as I'm at this step in the arduino LED blinker tutorial, I haven't added a component or anything 

https://github.com/fprime-community/fprime-tutorial-arduino-blinker/blob/main/docs/building-for-arduino.md#test-deployment

ethancheez, 2024-11-03:
> I wouldn't expect that as I'm at this step in the arduino LED blinker tutorial, I haven't added a component or anything

The default configuration in `fprime/config/` is even too large for the Teensy. Reference [this config file](https://github.com/fprime-community/fprime-tutorial-arduino-blinker/tree/main/config) for your project.

The big ones are:
- [TlmChanImplCfg.hpp](https://github.com/nasa/fprime/blob/devel/config/TlmChanImplCfg.hpp), `TLMCHAN_HASH_BUCKETS = 500`
- [CommandDispatcherImplCfg.hpp](https://github.com/nasa/fprime/blob/devel/config/CommandDispatcherImplCfg.hpp), `CMD_DISPATCHER_DISPATCH_TABLE_SIZE = 100` and `CMD_DISPATCHER_SEQUENCER_TABLE_SIZE = 25`
- [FpConfig.h](https://github.com/nasa/fprime/blob/devel/config/FpConfig.h#L324), `FW_TLM_BUFFER_MAX_SIZE`

ethancheez, 2024-11-03:
You can also download the [fprime-baremetal sizing utility](https://github.com/fprime-community/fprime-baremetal) for debugging of component/deployment/binary sizes. Instructions are in the README.