fprime-community/fprime-arduino #33: Arduino CLI wrapper support change breaks fprime-util generate

nateinaction, 2024-10-27:
Hey team,

I'm seeing the following error:
[image]

I have verified the board definition is installed:
```
arduino-cli board listall |grep rpipico
Raspberry Pi Pico                                                       rp2040:rp2040:rpipico
```

and I'm using a relatively recent version of `arduino-cli`
```
arduino-cli version
arduino-cli  Version: 1.0.4 Commit: Homebrew Date: 2024-08-12T13:32:50Z
```

It looks like there was an update to in `cmake/toolchain/support/arduino-wrapper.py` #30 that's causing this issue and I confirmed I could generate without seeing this error after reverting https://github.com/fprime-community/fprime-arduino/commit/1efa71306870c6b913186f14d051328ff9e711ab.

I noticed that there seems to be [some work-in-progress](https://github.com/SterlingPeet/arduino-cli-cmake-wrapper/pull/10) in the arduino-cli cmake wrapper repo. Is there a recommended version of the wrapper that I should be using other than what I get when I `pip install arduino-cli-cmake-wrapper`?

ethancheez, 2024-10-27:
> It looks like there was an update to in `cmake/toolchain/support/arduino-wrapper.py` #30 that's causing this issue and I confirmed I could generate without seeing this error after reverting [1efa713](https://github.com/fprime-community/fprime-arduino/commit/1efa71306870c6b913186f14d051328ff9e711ab).

Yeah this is the reason for the error. The PR has already been merged into main so I did not revert it, and all my Arduino OSAL work branched off of this commit. 

To solve this before the official release of `arudino-cli-cmake-wrapper` through `pip`, clone the [`arduino-cli-cmake-wrapper`](https://github.com/SterlingPeet/arduino-cli-cmake-wrapper/tree/refactor-tooling-cleanup) repo, change to the `refactor-tooling-cleanup` branch, and run this:
```sh
pip install .
```
> Make sure you are in your project virtual environment.

LeStarch, 2024-10-28:
@ethancheez yes, you are correct.  I will release an interim version of the tool today.

nateinaction, 2024-10-28:
> To solve this before the official release of arudino-cli-cmake-wrapper through pip, clone the [arduino-cli-cmake-wrapper](https://github.com/SterlingPeet/arduino-cli-cmake-wrapper/tree/refactor-tooling-cleanup) repo, change to the refactor-tooling-cleanup branch, and run this:

Thanks, this workaround worked for me!

ethancheez, 2024-10-30:
@nateinaction 
Pre-release `v0.2.0a1` is now available. Install with:
```sh
pip install arduino-cli-cmake-wrapper==0.2.0a1
```

closed 2024-11-01 by ethancheez