fprime-community/fprime-arduino #47: Returning something other than Drv::ByteStreamStatus::OP_OK can cause queue overflow issues

laboratory10, 2025-08-12:
https://github.com/fprime-community/fprime-arduino/blob/f86dfb47c66aea238eef56b5b53cfaf7021225aa/Arduino/Drv/StreamDriver/StreamDriver.cpp#L29-L31

By passing the return value of write_data back to whatever calls send_handler, it is possible to pass Drv::ByteStreamStatus::OP_OK or Drv::ByteStreamStatus::OTHER_ERROR. When other error is returned for me (in my case it is returned because my board tries to write data to serial but I haven't yet connected my computer to read the serial buffer), I quickly see an overflow in ComQueue with the WARNING_HI EVR QueueOverflow and all downlink stops entirely:

```
EVENT: (512) (2:4,298807) WARNING_HI: QueueOverflow : The COM_QUEUE (0) queue at index 1 overflowed
```

I'm currently addressing this by simply writing the data and returning OP_OK (which is essentially what you had before the v4 updates)

```
Drv::ByteStreamStatus StreamDriver ::send_handler(const FwIndexType portNum, Fw::Buffer& serBuffer) {
    void write_data(serBuffer);
    return Drv::ByteStreamStatus::OP_OK;
}
```

I understand not wanting to blindly return an OK status even if the write wasn't successful. If that doesn't sit well with you maybe you could try another option? Either finding a way to clear the queue even when the write isn't successful or maybe something like:

```
Drv::ByteStreamStatus StreamDriver ::send_handler(const FwIndexType portNum, Fw::Buffer& serBuffer) {
    Drv::ByteStreamStatus status =  write_data(serBuffer);
    if (status !=  Drv::ByteStreamStatus::OP_OK) {
        //issue a warning EVR?
    }
    return Drv::ByteStreamStatus::OP_OK;
}
```

ethancheez, 2025-08-12:
Good catch.. I am okay with always returning `Drv::ByteStreamStatus::OP_OK`. I don't see any issues if serial write errors are not handled.

ethancheez, 2025-08-12:
You are welcome to open a PR!

laboratory10, 2025-08-12:
Ok! I was hoping it would let me push a branch since I am in the fprime-community org but I don't appear to have the proper permission. Can you help with that?

PR #48 (2025-08-13): Prevent queue overflow if stream isn't being read
This pull request fixes #47
closed 2025-08-13 by ethancheez